siara-cc/esp_arduino_sqlite3_lib #1: (D) Can't open database: out of memory for ESP8266....

fred2088, 2018-10-18:
Sketch uses 589664 bytes (56%) of program storage space. Maximum is 1044464 bytes. Global variables use 68196 bytes (83%) of dynamic memory, leaving 13724 bytes for local variables. Maximum is 81920 bytes.

siara-cc, 2018-10-18:
Hi, the free memory is too low (13724 bytes).  Unfortunately SQLite requires lot of free heap and stack.  Please see the limitations (https://github.com/siara-cc/esp_arduino_sqlite3_lib/#limitations-on-esp8266). Please try the example code and db first and work from there. In my case, I can't make it work if free memory is less than 20k.

siara-cc, 2018-10-18:
Also, please check if your database page size is 512 bytes (using .dbinfo on command line of sqlite3).

closed 2018-11-12 by siara-cc