siara-cc/esp_arduino_sqlite3_lib #2: Support with ESP32 SDK 1.0.1rc version from board manager

tobozo, 2018-11-28:
Hello and thanks for this awesome library !

Although the project description mentions ESP8266, I've been playing with it on ESP32 for a few weeks now and it [works like a charm](https://github.com/tobozo/ESP32-BLECollector).

Last night Espressif did a release candidate for the 1.0.1 SDK for Arduino IDE (which you can update from your boards manager) and today the sqlite3 library won't compile.

<details>
  <summary> Compile log (...)</summary>

```
In file included from ~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:13476:0:
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'pcache1TruncateUnsafe':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:48838:11: error: 'nPage' undeclared (first use in this function)
   assert( nPage<0 || pCache->nPage==(unsigned)nPage );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                    ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:48838:11: note: each undeclared identifier is reported only once for each function it appears in
   assert( nPage<0 || pCache->nPage==(unsigned)nPage );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'pcache1Fetch':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:49201:11: error: 'pCache' undeclared (first use in this function)
   assert( pCache->bPurgeable || createFlag!=1 );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3PagerUnrefNotNull':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:55770:33: error: 'pPager' undeclared (first use in this function)
   assert( sqlite3PcacheRefCount(pPager->pPCache)>0 );
                                 ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'cellSizePtr':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:63969:18: error: 'debuginfo' undeclared (first use in this function)
   assert( nSize==debuginfo.nSize || CORRUPT_DB );
                  ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'cellSizePtrNoPayload':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:63990:11: error: 'debuginfo' undeclared (first use in this function)
   assert( debuginfo.nSize==(u16)(pIter - pCell) || CORRUPT_DB );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'balance':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:70758:17: error: 'balance_deeper_called' undeclared (first use in this function)
         assert( balance_deeper_called==0 );
                 ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3_backup_step':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:73439:15: error: 'rc2' undeclared (first use in this function)
       assert( rc2==SQLITE_OK );
               ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3Step':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:81073:64: error: 'Vdbe {aka struct Vdbe}' has no member named 'rcApp'
   assert( (p->rc!=SQLITE_ROW && p->rc!=SQLITE_DONE) || p->rc==p->rcApp );
                                                                ^
[... 134 lines cut ...]
               ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:106230:11: error: 'nLookaside' undeclared (first use in this function)
   assert( nLookaside==0 || nLookaside==sqlite3LookasideUsed(db,0) );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3InitCallback':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:122300:25: error: 'rcp' undeclared (first use in this function)
     assert( (rc&0xFF)==(rcp&0xFF) );
                         ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
```

</details>
<br><br>

Since I can easily rollback to the 1.0.0 and get a fully functional version this isn't a blocker for me, just leaving this issue here so anyone in the same situation can followup.

I'll be posting an issue on espressif tracker too.

siara-cc, 2018-11-28:
Hi, thank you for using the library.
FYI, there is a separate library for ESP32 here: https://github.com/siara-cc/esp32_arduino_sqlite3_lib/graphs/traffic
There are a lot of differences between SDK of esp8266 and esp32.  esp32 is more straightforward in terms of vfs implementation.  Please check it out.

siara-cc, 2018-11-28:
Actually I checked your page, it seems to be already using the library for esp32.  Are you still facing the issue?

tobozo, 2018-11-28:
whoops that's the library I'm using, I just posted on the wrong repo, should I close this a reopen on the other repo ?

siara-cc, 2018-11-28:
Yes. better for tracking i think..

closed 2018-11-28 by tobozo