siara-cc/esp_arduino_sqlite3_lib #3: NODEMCU IN SQLITE

kabilramar, 2019-04-22:
C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp: In function 'int encodeCount(char*, int, int)':

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:83:9: error: 'uint16_t' does not name a type

   const uint16_t adder[7] = {0, 4, 36, 164, 676, 4772,  0};

         ^

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:89:42: error: 'adder' was not declared in this scope

       ol = append_bits(out, ol, (count - adder[i]) << (16 - bit_len[i]), bit_len[i], 1);

                                          ^

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp: In function 'int readCount(const char*, int*, int)':

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:361:9: error: 'uint16_t' does not name a type

   const uint16_t adder[7] = {4, 0, 36, 164, 676, 4772,  0};

         ^

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:365:61: error: 'adder' was not declared in this scope

   int count = getNumFromBits(in, *bit_no_p, bit_len[idx]) + adder[idx];

                                                             ^

error: 'adder' was not declared in this scope

kabilramar, 2019-04-23:
I am using the same 1.9 and arduino core 2.5 but still i am not able to get rid of this issue

kabilramar, 2019-04-23:
Can u please like to share the code of how to create a table in sqlite3 using nodemcu

siara-cc, 2019-04-23:
> I am using the same 1.9 and arduino core 2.5 but still i am not able to get rid of this issue

Please download the repo and try again. Changes made to `/src/shox96_0_2.cpp`.

siara-cc, 2019-04-23:
> Can u please like to share the code of how to create a table in sqlite3 using nodemcu

Please refer to https://nodemcu.readthedocs.io/en/master/modules/sqlite3/
for sample code.

kabilramar, 2019-04-23:
Thank u for your immediate response

kabilramar, 2019-04-23:
Same error persists after download the repo file and now the errors are

 Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp: In function 'int encodeCount(char*, int, int)':

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:84:9: error: 'uint16_t' does not name a type

   const uint16_t adder[7] = {0, 4, 36, 164, 676, 4772,  0};

         ^

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:90:42: error: 'adder' was not declared in this scope

       ol = append_bits(out, ol, (count - adder[i]) << (16 - bit_len[i]), bit_len[i], 1);

                                          ^

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp: In function 'int readCount(const char*, int*, int)':

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:362:9: error: 'uint16_t' does not name a type

   const uint16_t adder[7] = {4, 0, 36, 164, 676, 4772,  0};

         ^

C:\Users\kabilesh.AURORA\Documents\Arduino\libraries\esp_arduino_sqlite3_lib-master\src\shox96_0_2.cpp:366:61: error: 'adder' was not declared in this scope

   int count = getNumFromBits(in, *bit_no_p, bit_len[idx]) + adder[idx];

                                                             ^

siara-cc, 2019-04-23:
Strange.. I tried with NodeMCU 0.9 board and I am able to compile successfully.
I do have a Windows 10 machine.  I will install Arduino, try and get back to you.

kabilramar, 2019-04-25:
Arun Ji can u able to  get the results in arduino

siara-cc, 2019-04-25:
I installed 1.8.9 in Win10, installed esp8266 arduino core 2.5.0, cloned from this repo, and it compiles successfully.  Please see the attached picture.

[image]

kabilramar, 2019-04-25:
Thank u ji for your response I will try and get back to you

kabilramar, 2019-04-26:
Thank u arun ji Finally it worked Thank u very much for your help and timely response

closed 2019-04-26 by kabilramar