siara-cc/esp_arduino_sqlite3_lib #4: cant open the database in esp8266 sqlite3

kabilramar, 2019-04-27:
I have copied the databases mdr512.db census2000names.db and placed inside the folder data.
When I run the program sqlite3_sdfat it throws the error Can't open database: Unable to open database".

**What may be the reason for that. Why it throws after I copied the database from the Github**

siara-cc, 2019-04-27:
If you are trying sdfat example, you need to manually copy the files to SD Card (to root folder).

siara-cc, 2019-04-27:
There are many limitations in using Sqlite db with ESP8266 especially because of RAM. Also stack size to be increased.  Please see README.md of this repo.

kabilramar, 2019-04-27:
I created a folder named data in the sd card and I manually copied the db files into the folder named data.

kabilramar, 2019-04-27:
This is the free space available information in mu arduino ide after sqlite_sdfat program was compiled.
 
Sketch uses 522648 bytes (50%) of program storage space. Maximum is 1044464 bytes.
Global variables use 47668 bytes (58%) of dynamic memory, leaving 34252 bytes for local variables. Maximum is 81920 bytes.

kabilramar, 2019-04-27:
I also created a folder named FLASH in the sd card and I manually copied the .db files (mdr512.db census2000names.db) into the folder named FLASH.

siara-cc, 2019-04-27:
The db files are to be copied to root folder.  I will mention it in the doc.
As for RAM, I was just pointing out because you may face a lot of problems when trying to write your own code.  The examples work fine because they are specifically designed to consume less RAM.

kabilramar, 2019-04-27:
Sir,

In the sqlite_sdfat program it was mentioned as" Before running please copy following files to SD Card:
    **data/mdr.db**". So I copied the files inside the folder named data. 

                               What is the root folder indicates in SD card.

kabilramar, 2019-04-27:
I am trying out the examples which you have uploaded like sqlite_sdfat,sqlite_smalldb.

siara-cc, 2019-04-27:
Please don't copy the files to any folder.  Just copy it to the main directory.  It is also called root folder.  If your SD Card is drive F, then use `copy data\mdr.db F:\`.

I understood the confusion.  Usually, the SPIFFS utility for ESP8266 copies the files in data folder to the root of SPIFFS.  Thats why I did not mention it specifically.  Now I will.

kabilramar, 2019-04-28:
Sir,
Is it necessary to increase the stack size in cores/esp8266/cont.h to at least 6144 (from 4096) when I run the examples program (sqlite3_sdfat)?. What is the reason to do that?

I am connecting the NODEMCU and Micro SD Card in the following Pin Correction. Is it right?

NODEMCU                      MICRO SD 
VIN                                   VCC
GND                                 GND
D8                                     CS
D5                                    SCK
D6                                    MISO
D7                                    MOSI

kabilramar, 2019-04-29:
arun ji are the connections mentioned above is correct??

siara-cc, 2019-04-29:
I am not sure about NodeMCU.  Please refer to their documentation.
Also see https://github.com/siara-cc/esp_arduino_sqlite3_lib/issues/5#issuecomment-487456337

siara-cc, 2019-04-29:
if the connections are different, you may have to change the code.

closed 2019-04-29 by kabilramar