siara-cc/esp_arduino_sqlite3_lib #5: SQL command cannot be executed - Error: File is encrypted or is not a database.

ts-abc, 2019-04-29:
I first tried the sdfat example, then the console_spiffs, as last the db_small example. All of them show the same Error, "file is encrypted or is not a database".

_Monitor output in last db_small:
`Opened database successfully`
`Failed to fetch data: file is encrypted or is not a database<br><br><a href='/'>back</a>`_

What can I try, better how do I solve this problem. Databases are definitely in order, all can be opened from the SD card and queried.

siara-cc, 2019-04-29:
It seems to be because of changes in the dependent library SdFat.  It works fine if I downgrade it to 1.0.7.

I intend to remove the dependency by using the standard SD library in future.

kabilramar, 2019-04-29:
Ji, whenever I download any db from the github (or) esp8266 sqlite3; In the properties of the db shows the following message as This file came from another computer and might be blocked to help protect this computer. 

"May be that be the reason for can't open the database"

ts-abc, 2019-04-29:
> 
> 
> It seems to be because of changes in the dependent library SdFat. It works fine if I downgrade it to 1.0.7.
> I intend to remove the dependency by using the standard SD library in future.
> Also don't forget to change the stack size to 6144 in `cont.h`

Oh gosh, correct, on the point, the downgrading to 1.0.7 fixes the problem completely. Yeah I had tested the cont.h before, from 7 or was it 8 bytes the ESP doesn't start anymore.

Yes the removing dependencies is a good idea. The only question is, it's possible without SD library. I don't know the details yet, I'm only working rigth now 3 weeks with the ESP's and co. But maybe it's possible to get the opened file handle, so every library could be used to access the file system.

ts-abc, 2019-04-29:
@kabilramar, I really don't understand what you want to say or ask, but is this the return error message of the SQL command or simply from the system's downloader. You should describing your problem in more detail and exclude any errors that might be caused by other thinks. First check if the SD card can be read or files can be opened.

kabilramar, 2019-04-30:
Yes man I could read or write text file in sdcard  interfacing nodemcu. 
But when I tried the sdfat program it says Unable to open Database what is the reason???
I changed by downgrading the sdfat library to 1.0.7.

kabilramar, 2019-04-30:
@ts-abc and I am using a Catalex Micro Crad Adapter instead of wemos Micro Shield

kabilramar, 2019-04-30:
@siara-cc is it necessary to use only Wemos sd shield and not catalex micro adapter

Auroratechive, 2019-04-30:
@siara-cc  I read you ESP32 in Hackster io It was so nice. This is the connection for ESP32 which i have referred in Hackster.io
SD Card    |  ESP32
*  DAT2 (1)      -
*  DAT3 (2)      SS (D5)
*  CMD  (3)      MOSI (D23)
*  VDD  (4)      3.3V
*  CLK  (5)      SCK (D19)
*  VSS  (6)      GND
*  DAT0 (7)      MISO (D18)
*  DAT1 (8)      -

**Can you please mention the wiring connection for esp8266 and micro SD card shield**

raghuram-bala, 2019-04-30:
Can I change mount(Sd0/Ss) to mount (sd0/8 )

raghuram-bala, 2019-04-30:
If you have free time please publish it as an video it will be more useful for easy understanding

siara-cc, 2019-04-30:
@kabilramar I am not sure about catalex adapter
@Auroratechive Please see the picture on README page. The connection is shown in the adapter itself.

> Can I change mount(Sd0/Ss) to mount (sd0/8 )

@raghuram-bala please see the example sdfat to find out how the SS pin can be changed

> If you have free time please publish it as an video it will be more useful for easy understanding

I may be a good idea to make a video. Presently I am unable to spare much time for this.

siara-cc, 2019-04-30:
I would like to point out that given the limited RAM on ESP8266 and even ESP32, it would be a challenge to get a project working, even if you manage to solve the connection issues.  It was difficult to get the examples working, especially on esp8266.  Only pagesize 512 would work that too with without row id option and good index design.  Insert frequency and Memory fragmentation is to be considered.
Now there is no doubt that they are working on the WeMos board with SdFat 1.0.7, but on any other board and any other use case, I may not be able to help much remotely and you are on your own.

raghuram-bala, 2019-05-01:
@ arun Thank you arun

kabilramar, 2019-05-01:
@ arun it means

D8 of the Wemos Mini Shield to be connected to SS pin on the SD Card Shield
D7 of the Wemos Mini Shield to be connected to MOSI pin on the SD Card Shield
D6 of the Wemos Mini Shield to be connected to MISO pin on the SD Card Shield
D5 of the Wemos Mini Shield to be connected to CLK pin on the SD Card Shield

"Is it right sir"

raghuram-bala, 2019-05-01:
@ arun 
Where should the following line should be given ?? On **cont.h** or some other .h files 

Please use a page size of 512 using the commands PRAGMA page_size=512; VACUUM;

Alacritous, 2019-09-14:
Is this still a thing? I'm getting this error. I've made all the suggested changes, including reverting the version of sdfat and editing the cont.h file and changing the page size to 512 and vacuuming the file with an SQLite editor on my computer.  The SD card is formatted with FAT32 and is a 16gb card. The sketch claims to be able to open the database file(I'm using the babyname.db file) but when I try the sql statement
 ```'select * from gendered_names where name like "m%";'  ```

and get the response above. 

I'm using the sketch sqlite3_console_sd.ino  (the only thing I've changed is the baudrate to 115200 which works fine)

The DirectoryFunctions.ino SD Card example sketch accurately reports the files on the SD card.

siara-cc, 2019-09-15:
The file gets corrupted sometimes. I assume it is because of power disconnection when write happens.  I am not sure.

I suggest copy the file again, try the example sketch first using the clone of the present repo and then modify from there.

Alacritous, 2019-09-15:
I was using a Wemos D1 Mini with an external SPI SD card reader. I gave up on that.  I have switched to an ESP32 TTGO T1 with an integrated SD card reader and switched to the SQLite3ESP32 library and it's working fine.   Faster too. I need to run 3-5 sequences that'll each be writing to the SD card about 500 times a second.  I hope it can keep up.