siara-cc/esp_arduino_sqlite3_lib #6: CAN'T OPE DATABASE IN SQLITE3 IN NODEMCU

gopalakartkikeyan, 2019-05-05:
CAN'T OPEN DATABASE IN SQLITE3 IN NODEMCU

gopalakartkikeyan, 2019-05-05:
I tied your example programs right now.
Here are the sample screenshots for your reference. Please check and tell me where I am mistaken.
 
 
[image]

gopalakartkikeyan, 2019-05-05:
The above image is the contents present in sd card. I have copied the database in the sd card

gopalakartkikeyan, 2019-05-05:
NEXT IMAGE is the error I am always getting when I run the program
[image]

gopalakartkikeyan, 2019-05-05:
location of the libraries which I
[image]
 have installed

gopalakartkikeyan, 2019-05-05:
next screenshot is esp8266 SDK location 
[image]

gopalakartkikeyan, 2019-05-05:
I could able to upload it successfully. here is the screenshot.
[image]

gopalakartkikeyan, 2019-05-05:
running in the nodemcu 1.0 dev kit.
[image]

gopalakartkikeyan, 2019-05-05:
Please tell me where i was mistaken. And my conctions are
D5->SCK
D6-> SO
D7-> MOSI
D8-> CS

gopalakartkikeyan, 2019-05-05:
I changed the stack size to 6144. here ist screenshot
[image]

gopalakartkikeyan, 2019-05-05:
I am using ARDUINO IDE 1.8.9 and espcore 2.5
[image]

gopalakartkikeyan, 2019-05-05:
I COULD ABLE TO WRITE & READ THE CONTENTS IN THE SD CARD USING SD LIBRARY. here the screenshot.

[image]

siara-cc, 2019-05-05:
> NEXT IMAGE is the error I am always getting when I run the program
> [image]

I appreciate all the details given.  There seems to be no problem with connectivity.  I suspect the problem could be because the SD Card supports only 8.3 filenames (such as CENSUS~1.DB).  Try changing the file name in the example ino exactly as it appears in the list.  Also check whether you are using SdFat library version 1.0.7.  There seems to be a problem with latest version of SdFat.

kabilramar, 2019-05-06:
I changed the name in the sdfat.ino as CENSUS~1.DB as the following screenshot. But I couldn't open the database. What may be the reason
[image]

gopalakartkikeyan, 2019-05-06:
I changed the name as CENSUS~1.DB in the sdfat.ino file but i couldn't open the database. what may be the reason.
 

[image]

gopalakartkikeyan, 2019-05-06:
Here is the change in sdfat.ino file
[image]

gopalakartkikeyan, 2019-05-06:
I am using sdfat 1.0.7 version here is the screenshot.
[image]

gopalakartkikeyan, 2019-05-06:
Why I am not get connected to the database ????? Is there any other things which I need to done. Please rectify this issue.

siara-cc, 2019-05-06:
I have tried the example on Win10 and it works.  You could try reformatting the SD Card with FAT32.
If still not working I will look into it during the weekend but its hard to find out what makes it not to work on your side.

gopalakartkikeyan, 2019-05-07:
Yes it is a FAT32 FILE SYSTEM. HERE IS THE SCREENSHOT.
[image]

gopalakartkikeyan, 2019-05-07:
I am getting the sqlite primary error code as 14 unable to open database file.

gopalakartkikeyan, 2019-05-07:
After reformatting here is the list of db files. I have the db files in root directory and inside the folder named DBS for testing purpose.

[image]

siara-cc, 2019-05-07:
If it is already FAT32, why is it not possible to retain the file name as it is?

> Here is the change in sdfat.ino file
> [image]

In this case you have modified as MDR512.db, but the filename is MDR512.DB.  Can this be the reason?

closed 2019-07-23 by siara-cc