siara-cc/esp_arduino_sqlite3_lib #8: Out of memory error

vignesh13404387, 2019-06-24:
Hello, I am using sqlite3 library in my program to store data on SD card using wemos D1 R1. When I compile the code it shows 60%of program storage space is used and the program throughs out of memory error.  When I searched the web it lead to add PRAGMA page_size 512.  I have no idea about where to add it. Please help.  Thankyou

siara-cc, 2019-06-25:
To change page size follow http://sqlite.1065341.n5.nabble.com/How-to-modify-page-size-td87692.html

vignesh13404387, 2019-06-25:
[image]

I uploaded this program 

[image]

And this is the output

vignesh13404387, 2019-06-25:
[image]
I uploaded this program and below is my output any suggestions please
[image]

siara-cc, 2019-06-25:
Support of `PRAGMA` is disabled in the library.  Page size is 512 default when you create using this library.  So need not set it explicitly.  Even then need to set it before opening the DB.
Please check:
- Are all the examples running without error.
- Have you changed stack size in `cont.h` to 6144
- Send code giving error if possible.

closed 2019-07-23 by siara-cc