siara-cc/esp_arduino_sqlite3_lib #9: SQL Can't open database from SPIFFS

aguilard07, 2019-07-27:
Hello, i'm trying to open a database of 36 Kbs with around 10 registers with no sucess in my NodeMCU V3.
I uploaded the .db file with "ESP8266 Skecth Data Upload" and with this code:

`Dir dir = SPIFFS.openDir("/");`
   `while (dir.next()) {`    
       `String fileName = dir.fileName();`
       `size_t fileSize = dir.fileSize();`
       `Serial.printf("FS File: %s, size: %ld\n", fileName.c_str(), (long) fileSize);`
   `}`
 I can visualize the file correctly but I always get the error: unable to open database.
I oh ready changed the stack size in cores/esp8266/cont.h
    to 6144 from 4096 and downgraded the SdFat library to version 1.07 but I've not been capable to consult the database.

siara-cc, 2019-07-27:
Which board you are using?  NodeMCU v3? 
Can't find it in my board list?  
[image]
Pls see attached picture.

aguilard07, 2019-07-27:
Sorry, my mistake. My board is NodeMCU V1.0(ESP-12E module)

siara-cc, 2019-07-28:
Hi, the small_db example is meant to be used with SD Card.  If you would like to use it with SPIFFS, have you changed the file path and initialised SPIFFS?

siara-cc, 2019-07-28:
`/FLASH/babyname.db` instead of '/SD0/babyname.db`

siara-cc, 2019-07-28:
Also need to include `<FS.h>` and set a file object reference like this:
```
File db_file_obj_1;
vfs_set_spiffs_file_obj(&db_file_obj_1);
```
before `openDb`.

Alternatively you could try the SPIFFS example to see if things are working first.

aguilard07, 2019-07-28:
I tried oh ready the SPIFFS example, all worked fine. But when I try to read a db that was before stored in the FLASH (like the babyname.db example) it can not opened it. 

I also included the `FS.h` file.

Yes  I put this code in my script: 
`File db_file_obj_1;`
`vfs_set_spiffs_file_obj(&db_file_obj_1);`

It must to be stored in the `/FLASH` directory, I mean I stored the db in the `/` directory and try opened it from there.

siara-cc, 2019-07-28:
Can paste your code here? If you mention /FLASH in openDb it opens from / (root folder).

aguilard07, 2019-07-28:
This is the code. The babyname.db is stored in `data`

`#include <stdio.h>`
`#include <stdlib.h>`
`#include <sqlite3.h>`
`#include <vfs.h>`
`#include <SPI.h>`
`#include <FS.h>`
`extern "C" {`
`#include "user_interface.h"`
`}`
`#include <ESP8266WiFi.h>`

`void WiFiOff() {`
   `wifi_station_disconnect();`
   `wifi_set_opmode(NULL_MODE);`
   `wifi_set_sleep_type(MODEM_SLEEP_T);`
   `wifi_fpm_open();`
   `wifi_fpm_do_sleep(0xFFFFFFF);`
`}`

`const char* data = "Callback function called";`
`static int callback(void *data, int argc, char **argv, char **azColName) {`
   `int i;`
   `Serial.printf("%s: ", (const char*)data);`
   `for (i = 0; i<argc; i++){`
       `Serial.printf("%s = %s\n", azColName[i], argv[i] ? argv[i] : "NULL");`
   `}`
   `Serial.printf("\n");`
   `return 0;`
`}`

`int db_open(const char *filename, sqlite3 **db) {`
   `int rc = sqlite3_open(filename, db);`
   `if (rc) {`
       `Serial.printf("Can't open database: %s\n", sqlite3_errmsg(*db));`
       `return rc;`
   `} else {`
       `Serial.printf("Opened database successfully\n");`
   `}`
   `return rc;`
`}`

`char *zErrMsg = 0;`
`int db_exec(sqlite3 *db, const char *sql) {`
  ` Serial.println(sql);`
   `long start = micros();`
   `int rc = sqlite3_exec(db, sql, callback, (void*)data, &zErrMsg);`
   `if (rc != SQLITE_OK) {`
       `Serial.printf("SQL error: %s\n", zErrMsg);`
       `sqlite3_free(zErrMsg);`
   `} else {`
       `Serial.printf("Operation done successfully\n");`
   `}`
  `Serial.print(F("Time taken:"));`
   `Serial.println(micros()-start);`
   `return rc;`
`}`

`void setup() {`

   `Serial.begin(115200);`
   `sqlite3 *db1;`
   `int rc;`

   `system_update_cpu_freq(SYS_CPU_160MHZ);`
   `WiFiOff();`

   `if (!SPIFFS.begin()) {`
      ` Serial.println("Failed to mount file system");`
    `   return;`
   `}`

   `// list SPIFFS contents`
   `Dir dir = SPIFFS.openDir("/");`
   `while (dir.next()) {    `
       `String fileName = dir.fileName();`
       `size_t fileSize = dir.fileSize();`
    ` Serial.printf("FS File: %s, size: %ld\n", fileName.c_str(), (long) fileSize);`
   `}`
   `Serial.printf("\n");`

   
   `sqlite3_initialize();`

  `// Open databases`
   `File db_file_obj_1;`
  ` vfs_set_spiffs_file_obj(&db_file_obj_1);`
   `if (db_open("/babyname.db", &db1))`
       `return;`
 
   
 `  sqlite3_close(db1);`

`}`

`void loop() {`
`}`

siara-cc, 2019-07-28:
Please try `db_open("/FLASH/babyname.db", &db1)`.  In this VFS, `FLASH` indicates SPIFFS and `SD0` indicates SD Card on SPI bus.

aguilard07, 2019-07-29:
OK, I changed the directory as you said, now I have this error:
`SQL error: file is encrypted or is not a database`

aguilard07, 2019-07-29:
Thanks for your help, I solved the issue.

siara-cc, 2019-07-29:
ok,  great..

aguilard07, 2019-07-30:
Hey, I'm noob using NodeMCU and these kind of uC. Could you please teach me how and where to use this command?
`PRAGMA page_size=512; VACUUM;`
Because I'm having the trouble "Out of memory"

siara-cc, 2019-07-31:
The given `babyname.db` already is created using `page_size=512`. And if you are creating a database using the library itself, it already creates using `page_size=512`.

So if you are getting `Out of memory`, it could be because of some other reason.  Sqlite3 is a bit of overkill for ESP8266 and `Out of memory` can be expected quite often if you don't design for it.

If you are creating on the desktop, you can open the db file using `sqlite3.exe` and use the above two commands.

closed 2019-08-13 by siara-cc