siara-cc/esp_arduino_sqlite3_lib #10: Can't compile the sqlite_spiffs example

orlo11, 2019-09-08:
Hello,
when I try to compile the sqlite_spiffs example, I get the following error:
`sqlite3_spiffs:48:4: error: 'System' was not declared in this scope
    System.out.println(sql);
    ^
Bibliothek Sqlite3 in Version 0.9 im Ordner: /home/thomas/Arduino/libraries/Sqlite3  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: /home/thomas/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/SPI  wird verwendet
Bibliothek ESP8266WiFi in Version 1.0 im Ordner: /home/thomas/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi  wird verwendet
Bibliothek SdFat in Version 1.0.7 im Ordner: /home/thomas/Arduino/libraries/SdFat  wird verwendet
exit status 1
...`
I use arduino 1.8.9 and downgraded the esp8266 sdk to 2.5.0 as you wrote earlier.
You know any help?
Thanks,

orlo11, 2019-09-08:
Sorry,
I noticed just now, that my version (I installed from the library-manager inside the IDE) seems to be outdated -- here in the repo the error seems to be already corrected...

closed 2019-09-08 by orlo11

siara-cc, 2019-09-08:
Hi, Thank you for informing me. I will check why the older version is in the library manager.