siara-cc/esp_arduino_sqlite3_lib #12: Questions from the core

devyte, 2019-10-29:
This is not an issue, feel free to close right away.

Hi, I'm a maintainer of the esp8266 Arduino core. I'm curious about this implementation, so I have some questions (apologies for asking here).

1. What is the status of this project? Still being maintained, will be maintained in the future, etc?

2. I see a refrence to core v2.0.0 which is beyond ancient. The latest core release is 2.5.2, and 2.6.0 is imminent. There have been large efforts to increase free available heap since 2.4.x. Have recent core versions been tested?

3. I see only the C api (official one). Will there be any effort to include any C++ wrappers? Not necessarily here, I'm thinking along the lines of "this works with C++ bindings xyz which can be found over there".

4. I saw the comments about the stack size. Have any recent tests been done? I ask because there have been stack size optimizations done, where the CONT stack is placed on top of the SYS (i.e.: sdk context) stack to release more heap. The optimization relies on a 4KB cont stack size, so I have to wonder what happens with the optimization and your 6KB+ requirement.

siara-cc, 2019-10-30:
Thank you for the feedback.  I do spend time on these projects, but on weekends only.  I tested it on 2.5.0, but could not find any significant difference.  SQLite consumes lots of stack and heap so I didn't look too closely.

Is the C++ requirement just for semantic reasons or more than that?  SQLite has a decent C api and as you know it works with C++ too.

I intend to upgrade the SQLite version and remove SdFat dependency soon.  I will go through your points in detail later when I find time.

I also have something that consumes far lesser memory in the offing.

Thanks.

devyte, 2019-10-30:
Thank you for you answer! 

> C++ requirement

Mostly just curious, I've only ever used the C api before. Our build supports enabling of C++ exceptions and whenever I'm faced with a C api I end up wondering about C++ bindings for scoped resource safety.

> tested it on 2.5.0

I strongly suggest keeping up with the latest release. Right now 2.6.0 is imminent, and several critical stability fixes will go into it, as well as mem usage changes. 

> SdFat dependency

The filesystem classes have been unified behind a single FS interface. That means that a FS object can serve as interface to SD FAT, or SPIFFS, or LittleFS (available alternative to spiffs for on flash FS). I suggest migrating to use a FS and its related File etc objects as interface. That would allow all possibilities like e. g. a db on flash.

I'm closing to reduce not count as open :)

closed 2019-10-30 by devyte