siara-cc/esp_arduino_sqlite3_lib #13: Error Compile ESP 8266 CORE 2.6.3

laercionit, 2020-02-24:
Hello, I'm having trouble compiling the example (sqlite3_sdfat) using ESP8266 on ArduinoIDE with core 2..6.3. someone with the same problem?

siara-cc, 2020-03-08:
Hi, I just upgraded to 2.6.3 and tried.. I am able to compile and run it.
You probably may have to get the SdFat library which is causing this problem.
Be sure to get 1.0.7 of SdFat because the latest version causes trouble. See https://github.com/siara-cc/esp_arduino_sqlite3_lib/issues/5#issuecomment-487456337

laercionit, 2020-03-08:
> See #5 (comment)

Core ESP 2.6.3 integrated SDFAT into its core, it now uses the latest
version. I had problems here, it does not compile if used together with FS.
I think we need to work on SQLite to support the latest version of SDFAT.

*Laercio Junior*
Cel./Pessoal: (21) 99162-7440
Email Pessoal: laercionit@gmail.com
Skype: laercionit

laercionit, 2020-03-09:
In this example I am using SPIFFS, which is more secure.

Using the ESP8266 Nodemcu 12-E with CORE 2.6.3
I am getting the errors below to compile.

**I do not even know where to begin with. Today I use EDB as a database in SD, I want to replace it with SQLite in SD.**

```

#include <stdio.h>
#include <stdlib.h>
#include <sqlite3.h>
#include <vfs.h>
#include <SPI.h>
#include <FS.h>
extern "C" {
#include "user_interface.h"
}
#include <ESP8266WiFi.h>

const char* data = "Callback function called";
static int callback(void *data, int argc, char **argv, char **azColName){
   int i;
   Serial.printf("%s: ", (const char*)data);
   for (i = 0; i<argc; i++){
       Serial.printf("%s = %s\n", azColName[i], argv[i] ? argv[i] : "NULL");
   }
   Serial.printf("\n");
   return 0;
}

int openDb(char *filename, sqlite3 **db) {
   int rc = sqlite3_open(filename, db);
   if (rc) {
       Serial.printf("Can't open database: %s\n", sqlite3_errmsg(*db));
       return rc;
   } else {
       Serial.printf("Opened database successfully\n");
   }
   return rc;
}

char *zErrMsg = 0;
int db_exec(sqlite3 *db, const char *sql) {
   Serial.println(sql);
   long start = micros();
   int rc = sqlite3_exec(db, sql, callback, (void*)data, &zErrMsg);
   if (rc != SQLITE_OK) {
       Serial.printf("SQL error: %s\n", zErrMsg);
       sqlite3_free(zErrMsg);
   } else {
       Serial.printf("Operation done successfully\n");
   }
   Serial.print(F("Time taken:"));
   Serial.println(micros()-start);
   return rc;
}

void setup() {
   Serial.begin(115200);
   sqlite3 *db1;
   char *zErrMsg = 0;
   int rc;
   char *sql;

   system_update_cpu_freq(SYS_CPU_160MHZ);

   SPI.begin();
   vfs_mount("/SD0", SS);

   sqlite3_initialize();

   // Open database 1
   if (openDb("/SD0/census2000names.db", &db1))  return;
   
   rc = db_exec(db1, "Select * from surnames where name = 'MICHELLE'");
   if (rc != SQLITE_OK) {
       sqlite3_close(db1);
       sqlite3_close(db2);
       return;
   }
   rc = db_exec(db1, "Select * from surnames where name = 'SPRINGER'");
   if (rc != SQLITE_OK) {
       sqlite3_close(db1);
       sqlite3_close(db2);
       return;
   }
   
   sqlite3_close(db1);
   

}

void loop() {
}
```

siara-cc, 2020-03-09:
Like I said, I think temporarily you can get it working with SdFat 1.0.7 downloaded using Library Manager till I fix it?

laercionit, 2020-03-09:
@siara-cc 

Unfortunately this is not possible.
In CORE 2.6.3 of ESP8266 (Arduino) he incorporated SDFAT, this is in the core of esp8266, there is no way to downgrade.

So I looked for help to try to put SQLite to run on this core, it would be a great replacement for EDB (SLOW)

bhullar89, 2023-09-08:
Is this issue resolved, as i am still getting below errors

`C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:6:8: error: 'SdFat' does not name a type
    6 | static SdFat *sdFats[] = {NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL};
      |        ^~~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp: In function 'vfs_vol* sdfat_mount(const char*, int)':
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:15:5: error: 'SdFat' was not declared in this scope; did you mean 'sdfat::SdFat'?
   15 |     SdFat *sdFat = new SdFat();
      |     ^~~~~
      |     sdfat::SdFat
In file included from C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:3:
C:\Users\pushE\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266SdFat\src/SdFat.h:402:17: note: 'sdfat::SdFat' declared here
  402 | typedef SdFat32 SdFat;
      |                 ^~~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:15:12: error: 'sdFat' was not declared in this scope; did you mean 'sdfat'?
   15 |     SdFat *sdFat = new SdFat();
      |            ^~~~~
      |            sdfat
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:15:24: error: expected type-specifier before 'SdFat'
   15 |     SdFat *sdFat = new SdFat();
      |                        ^~~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:18:5: error: 'sdFats' was not declared in this scope; did you mean 'sdfat'?
   18 |     sdFats[sd_idx] = sdFat;
      |     ^~~~~~
      |     sdfat
In file included from C:\Users\pushE\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266SdFat\src/ExFatLib/ExFatFile.h:36,
                 from C:\Users\pushE\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266SdFat\src/ExFatLib/ExFatVolume.h:28,
                 from C:\Users\pushE\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266SdFat\src/ExFatLib/ExFatLib.h:27,
                 from C:\Users\pushE\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266SdFat\src/SdFat.h:33,
                 from C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:3:
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp: In function 'vfs_file* sdfat_open(const char*, const char*)':
c:\users\pushe\onedrive\documents\arduinodata\packages\esp8266\hardware\esp8266\3.0.0\libraries\esp8266sdfat\src\common\fsapiconstants.h:122:19: error: 'O_RDONLY' was not declared in this scope; did you mean 'sdfat::O_RDONLY'?
  122 | #define O_READ    O_RDONLY
      |                   ^~~~~~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:36:18: note: in expansion of macro 'O_READ'
   36 |         imode |= O_READ;
      |                  ^~~~~~
c:\users\pushe\onedrive\documents\arduinodata\packages\esp8266\hardware\esp8266\3.0.0\libraries\esp8266sdfat\src\common\fsapiconstants.h:76:15: note: 'sdfat::O_RDONLY' declared here
   76 | const uint8_t O_RDONLY = O_READ;
      |               ^~~~~~~~
c:\users\pushe\onedrive\documents\arduinodata\packages\esp8266\hardware\esp8266\3.0.0\libraries\esp8266sdfat\src\common\fsapiconstants.h:123:19: error: 'O_WRONLY' was not declared in this scope; did you mean 'sdfat::O_WRONLY'?
  123 | #define O_WRITE   O_WRONLY
      |                   ^~~~~~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:38:21: note: in expansion of macro 'O_WRITE'
   38 |            imode |= O_WRITE;
      |                     ^~~~~~~
c:\users\pushe\onedrive\documents\arduinodata\packages\esp8266\hardware\esp8266\3.0.0\libraries\esp8266sdfat\src\common\fsapiconstants.h:80:15: note: 'sdfat::O_WRONLY' declared here
   80 | const uint8_t O_WRONLY = O_WRITE;
      |               ^~~~~~~~
c:\users\pushe\onedrive\documents\arduinodata\packages\esp8266\hardware\esp8266\3.0.0\libraries\esp8266sdfat\src\common\fsapiconstants.h:123:19: error: 'O_WRONLY' was not declared in this scope; did you mean 'sdfat::O_WRONLY'?
  123 | #define O_WRITE   O_WRONLY
      |                   ^~~~~~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:41:18: note: in expansion of macro 'O_WRITE'
   41 |         imode |= O_WRITE;
      |                  ^~~~~~~
c:\users\pushe\onedrive\documents\arduinodata\packages\esp8266\hardware\esp8266\3.0.0\libraries\esp8266sdfat\src\common\fsapiconstants.h:80:15: note: 'sdfat::O_WRONLY' declared here
   80 | const uint8_t O_WRONLY = O_WRITE;
      |               ^~~~~~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:43:22: error: 'O_CREAT' was not declared in this scope; did you mean 'sdfat::O_CREAT'?
   43 |             imode |= O_CREAT;
      |                      ^~~~~~~
      |                      sdfat::O_CREAT
In file included from C:\Users\pushE\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266SdFat\src/ExFatLib/ExFatFile.h:36,
[... cut ...]
      |                           ^~~~
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp: In function 'uint32_t sdfat_size(vfs_file*)':
C:\Users\pushE\OneDrive\Documents\Arduino\libraries\Sqlite3_for_ESP8266\src\sdfat_fns.cpp:123:34: error: 'class fs::File' has no member named 'fileSize'
  123 |   return ((File *)fd->file_obj)->fileSize();
      |                                  ^~~~~~~~
exit status 1
Error compiling for board Generic ESP8266 Module.
Board at COM5 is not available
`

siara-cc, 2023-09-09:
I think SdFat is not installed.  Please download and installed the library v1.0.7 from https://github.com/greiman/SdFat and try compiling again

bhullar89, 2023-09-11:
@siara-cc  is sdfat 2.2.2 ( latest ) version not supported ? Do we have any plans to support that ?

siara-cc, 2023-09-12:
@bhullar89 There was some issue quite some time back with a later version. I don't remember what the issue was.  If that issue has been fixed with sdfat 2.2.2, it may work. If you find it working please let me know so I can remove the note on restriction :-)

Elidlt, 2023-12-05:
> I think SdFat is not installed. Please download and installed the library v1.0.7 from https://github.com/greiman/SdFat and try compiling again

i had the same problem and install SdFat v 1.0.7 and My problem Solved.
by the way for those wanna use inside flash memory insted of  sdcard  This will confusing
its be good if seprate sd users from flash users.
thanks.