siara-cc/esp_arduino_sqlite3_lib #16: Out of memory after 65 inserts

basementmedia2, 2020-12-29:
Hi,

sqlite works now with my esp8266 NodeMCU thanx to your library.
But the following code results in a "out of memory" message after approx. 65 dataset inserts:

```
randNumber=random(50);

   for (int i=1;i<1000;i=i+1) {
      char query[200];
      sprintf(query, "Insert into messungen (datum, liter) VALUES ('2020-12-20 20:59:00', %d);", randNumber);
      rc = db_exec(db1, query);
      if (rc != SQLITE_OK) {
         sqlite3_close(db1);
         return;
      }
      delay(100);
   }
```
I've set the stack size to 6144 in the following file:
/home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/cont.h

Is there more i can try to get rid of the out of memory error?

Best wishes
Daniel

siara-cc, 2021-01-01:
What is the page size?  For a low memory MCU like Sqlite, it does not work well if page size > 512

basementmedia2, 2021-01-04:
Hi, where/how can i set the page size? I am using SQLite Studio.