siara-cc/esp_arduino_sqlite3_lib #21: Can't open database: Unable to open the database file  Error 14

gnithyalakshmi, 2022-05-27:
I have used your library and able to compile successfully in Sketch.  

On uploading to NODE MCU, only the sample with flash memory works.  But for the SD card option, it throws error as 'cant open database...unable to open database file.

Any help would be appreciated. 

I am using Node MCU ESP826 and my SD card is FAT32

siara-cc, 2022-05-28:
Usually it would be an issue with connection to SD Card.  Please see if the SD samples work before trying this lib.  Also please see the README for specific version of SD Fat library to be used.

gnithyalakshmi, 2022-05-28:
Thanks for quick response.

I have used the SDFAT 1.0.7 as you mentioned.  When I try to run the
examples of SDFAT library , everything fails except quickstart.  The
difference is opening file SD.begin with clock speed.  If I run the example
with #define SPI_SPEED SD_SCK_MHZ(4) then it works otherwise it fails.

Question to you is how should I change your code to include this clock
speed. Have you defined anywhere when you open a file with SD.begin ? I am
checking your source now but if you can help me quickly, it would be
great.

Thanks,
Nithya

siara-cc, 2022-05-28:
The calls to SDFat are in `sdfat_fns.cpp`, but I am not sure if it is needed to specify a clock speed.  If you see the picture in readme.md, I used a readymade shield for SD SPI connection and I did not have to modify the clock speed.  Perhaps you could see if there is some loose connection or if a resistor/capacitor needs to be added to support higher speeds.