siara-cc/esp_arduino_sqlite3_lib #23: vfs_mount not working, but regular SD communication is.

jnegrete2005, 2023-01-04:
I was using the library with no problems when suddenly, I got an error saying `unable to open database file`. I thought I was doing something wrong with the code or the wiring, but when I tried to use the SD with SD.h, it did open, so it is the function.  
I found out that it returns NULL in case of an error, and indeed it does, but I haven't changed the code or the wiring, so I don't know what the error is.

This is my code:

```CPP
void setup(void) {
  Serial.begin(115200);
  Wire.begin();
  RTC.begin();
  SPI.begin();
  nfc.begin();

  if (vfs_mount("/SD0", D8) == NULL)
    Serial.println("Unable to set vfs");

  if (RTC.lostPower())
    RTC.adjust(DateTime(F(__DATE__), F(__TIME__)));

  sqlite3_initialize();

  Serial.println("System initialized");
}
```

And on behalf of the SD, I am wiring:
- D5 → `SCK`
- D6 → `MISO`
- D7 → `MOSI`
- D8 → `CS` 

I hope I am doing something wrong to fix it quickly.

Thanks for any help.

siara-cc, 2023-01-04:
Hi, sounds like there could be some noise in the connections.
Try lowering the SPI speed (optional parameter to begin()).
Please also use shorter wires for the above connections.
If possible use a PCB with ground planes covering these lines.

jnegrete2005, 2023-01-04:
Hi! Thanks for the reply.
I am using the shortest jumper wires available, I even changed them.
How do I lower the speed? It says that `SPI.begin()` takes no parameters, so it is not from there. If you need any more of my code, I will provide it.

siara-cc, 2023-01-06:
Hi, please see this for changing clock speed:
https://github.com/siara-cc/esp_arduino_sqlite3_lib/issues/21#issuecomment-1140217727
I have not tried it, but I know that higher speeds often cause issues.  You may have to wade through docs to find out how to set it, if that is causing your issue.  I am not sure what else could be the problem.
Another potential issue is number of files open which also I remember can be set when initialising SPIFFS/LITTLEFFS.

jnegrete2005, 2023-01-06:
Nope, it is still not working. I reestarted the project, but it is still not working.

jnegrete2005, 2023-01-06:
Maybe I can go buy an esp32 and try with that, or change the esp8266. What do you think is the best option? Cause if it suddenly stopped working, it must be something with the hardware, and I can't determine what it is.

jnegrete2005, 2023-01-07:
I found another way to lower the speed and indeed it worked. I had to go to `.SdFat\src\SdCard\SdInfo.h`, to check the speed definitions. `FULL` and `DIV3` speeds don't work, I had to copy `SPI_HALF_SPEED`. Then paste it `SdFat\src\SdFat.h` in the functions `begin` and `cardBegin`, replacing the `SPI_FULL_SPEED` in the parameters of the functions (you can find these by CTRL + Click in `begin` in `if (!sdFat->begin(num))` inside `Sqlite3 for ESP8266\src\sdfat_fns.cpp`). 

Thank you for your time.

closed 2023-01-07 by jnegrete2005

siara-cc, 2023-01-08:
Great!  I have faced this issue before quite some time ago and thats why I could related to your issue.

Wires or even PCB lines pick up noises from nearby lines or static unless shielded with Ground.  So with faster speeds the detection of 0 or 1 for digital communication is more erratic because of the noises.  Electromagnetic noises cause less interference at lower speeds as the MCU has more time to detect 0 or 1.  The second problem is that jumper wires are highly prone to loose contact.

So it is always better to make a PCB instead of using wires for connection and there are rules to how much gap should be present between lines and where possible make sure the lines are surrounded by Ground planes with vias to other side.