siara-cc/esp_arduino_sqlite3_lib #24: sqlite3_spiffs.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

AmirAmraei, 2023-01-06:
hi
i have arduino 1.8.19 and install esp8266 3.0.2
but cant compile 
I get the following error

c:/users/developer/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\DEVELO~1\AppData\Local\Temp\arduino_build_390221/sqlite3_spiffs.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

why?

siara-cc, 2023-01-07:
Hi, I please check the settings in the menu related to size allocated to Program size and SPIFFS size.

AmirAmraei, 2023-01-07:
tnk
So at least 160 kilobytes of spiffs are needed
I tested this

closed 2023-01-16 by AmirAmraei